whitecatboard/Lua-RTOS-ESP32 #403: posix wite() failes..

mgliewe, 2022-04-11:
For some reason, the write(2) calls are failing on my build, resulting in ENOENT errors on each call.

This envolves the buildin ``edit()`` and dropbear not able to create its needed support files.

I temporally fixed that by patching both to use stdio calls (which are working fine), but i'm sure i was just curing the symptoms..

the0ne, 2022-04-12:
No C-code reference.
No Lua code.
No idea what you're talking about.

mgliewe, 2022-04-12:
sorry to have been so brief and confused, it was a long night... 
The problem was (is) that posix write() calls in C on the filesystem are failing with ENOENT, whereas stdio fwrite() calls succeed. That only applies to write() calls, open()/read() et al calls are working.

See https://github.com/mgliewe/Lua-RTOS-ESP32/commit/c0383af8ed16350f50ef3bf285a93f2046586e4c how i worked around that. I assume thats a problem i have with my toolchain, but i thought i should report it anyway as it had cost me some headache

the0ne, 2022-04-16:
Thanks for the link, now I understand what you mean.
That issue hasn't occurred on my system.
If I ever see it, I'll of cause let you know the solution 👍

mgliewe, 2022-05-31:
since I was the only one havin this issue this can go to waste?

closed 2022-05-31 by mgliewe