whitecatboard/Lua-RTOS-ESP32 #404: buildin ssh wishlist....

mgliewe, 2022-04-11:
I was pretty amazed to finally be able to connect via ssh to my board.. WTF, thats the f**ng wa we want it.

On my wishlist:
- more than one connection
- proper scp support
- puplic key auth

the0ne, 2022-04-12:
See CONFIG_LUA_RTOS_LUA_USE_SCP_NET
It's the scp _client_ only, but the direction wouldn't matter too much, right?

Here's how to use it:
```
net.scp.get("your_domain.com",22,"/server_path/server_file","/local_path/local_file","user","pass")
net.scp.put("your_domain.com",22,"/local_path/local_file","/server_path/server_file","user","pass")
net.ssh.exec("your_domain.com",22,"/usr/bin/your_command and args","user","pass")
```

mgliewe, 2022-05-02:
> - proper scp support

after some research i realized that scp isn't really what i was aiming at, my intention was rather being able to sshmount an esp board.

Alas, dropbear doesn't suport the sftp protocol by its own, but relies on an sftp-server binary laying around in the OS to do the work.

So, to accomplish this point on the wishlist, one had to implement the sftp protocol inside the dropbear context. 

On the bright sight, the protocol isn't to complicated in though of memory/resource constraints, and there are reference/example implementations one could adopt in lie of GPL.   

The cyclone [1] lib came up one my research, which seemed to bee suitable for adoption due to its examplary well designed capsulation of function and OS dependencies...

lg maG

[1] https://www.oryx-embedded.com/products/CycloneSSH.html