whitecatboard/Lua-RTOS-ESP32 #405: ldo support..

mgliewe, 2022-04-11:
micropython supports (limited) loadable code. 

mpy modules, compiled on host, may contain native code, loaded and relocated on runtime.

This could be a major feature for luartos as well? a major security issue as well, i suppose.

the0ne, 2022-04-14:
try this:
```
	local cmd = "your command here"
	print("now executing "..cmd)
	local function dostring (x) return assert(load(x), "")() end
	print(dostring(cmd))
```

closed 2022-04-14 by the0ne

mgliewe, 2022-04-14:
a misunderstanding... i was refering to native code.

My usecase: i'm working on audio dsp processing. Its a modular system. I have some base modules (i2s audio source/sink, BT/a2dp yet to come) and a bunch of dsp modules, which can dynamically  instantiated and routed.
These dsp modules are generated by faust dsp [1], which takes a description of an audio filter in a functional language, and translates that into c code. These dsp modules might be effect filters, equalizers, echo, distortion... you name it. 
Basically the system is set to work simillar to guitarix [2] on linux, on a very limited scale.

Due to limited horsepower of the xtensa cores, tweaking and working on the dsp filters is most of the work. The turn-around cycles for esp code are a bit lengthly (work on source - rebuild - reflash - reboot), experimenting with c code tends to be a bit anoying.

Micropython addresses this by means of a custom linker [3], taking .o files in ELF format, linking them into a proprietary ldo format, and corresponding loader code inside the python core [4].

[1] https://faust.grame.fr/
[2] https://guitarix.org/
[3] https://github.com/micropython/micropython/blob/master/tools/mpy_ld.py
[4] https://github.com/micropython/micropython/blob/master/py/persistentcode.c#L198

the0ne, 2022-04-14:
My hint would be to check out the VM functions. I never used that and I don't know much about it, but if it's what I think then it might help you.