whitecatboard/Lua-RTOS-ESP32 #407: How to build Lua RTOS with the screen module

sethpeace, 2022-05-02:
I see from [this wiki page](https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/SCREEN-module) that there is a screen module which i really want to use for my project. However, no prebuilt board OSes include support and there doesn't appear to be an option for it in the build settings. So how do you access it?

sethpeace, 2022-05-02:
On a related note, is there any chance the module will support more chipsets in the future?

sethpeace, 2022-05-03:
`make menuconfig` has no screen module option:
[image]

the0ne, 2022-05-03:
Maybe check the following, seems the wiki was not renamed:
https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/GDISPLAY-module

sethpeace, 2022-05-03:
you just saved my life thank you :)

closed 2022-05-03 by sethpeace