whitecatboard/Lua-RTOS-ESP32 #411: Is Lua-RTOS-ESP32 installed with python2 only?

burtonkent, 2022-05-21:
Been trying to make my own version of Lua with FastLED and Simplex callable C functions.  I was trying to simply build my own Lua as detailed at https://github.com/whitecatboard/Lua-RTOS-ESP32#method-2-build-by-yourself first, then worry about adding C modules.

The `make fresh` worked a little but kept complaining that pyparsing was out of date, so I tried:
`pip install -Iv pyparsing==2.0.3`
It still complained about the version.
Installed pyenv and made python 3 my default on my Macbook. Restarted the build, now `make fresh` fails with `NameError: name 'raw_input' is not defined`.

Am I right that this will only build with Python 2? And if so how do I fix the pyparsing problem? Thanks.

burtonkent, 2022-05-21:
Got it to work by using pyenv and redoing the whole install. I went ahead and created [a pull request](https://github.com/whitecatboard/Lua-RTOS-ESP32/pull/412) updating README.md to explain this requirement. Will close this issue if/when the pull request is merged.

the0ne, 2022-07-03:
Has been merged, thanks a lot!

closed 2022-07-03 by the0ne