whitecatboard/Lua-RTOS-ESP32 #413: Undeclared variable makes the OTA update compile fail.

burtonkent, 2022-05-21:
Steps to reproduce if you already have Lua-RTOS-ESP32 set up to 'make':
1. `make clean` to clear out past makes.
2. `make`
3. choose 8 (Espressif Systems ESP-WROVER-KIT)
4. choose 2 (Espressif Systems ESP-WROVER-KIT with OTA)
5. Will warn about not being able to remove directories because they're not empty. It's fine.
6. Fails in `components/sys/drivers/net.c` because the variable **`total`** isn't declared before use on line 513.
The error given is:
```
Lua-RTOS-ESP32/components/sys/drivers/net.c: In function 'net_ota':
<development path I'm using>/Lua-RTOS-ESP32/components/sys/drivers/net.c:513:78: error: 'total' undeclared (first use in this function)
                 int pct = (int)(((float)(address-update_partition->address))/total*100);
```
It's strange because git blame shows:
b22ddc939 components/sys/drivers/net.c      (the0ne       2020-01-09 07:43:20 +0100 513)                 int pct = (int)(((float)(address-update_partition->address))/*100); 
Note the 2020 commit date. I'm surprised this wasn't found as a bug before. I have to wonder if this is my error somehow.

burtonkent, 2022-05-21:
Also I did try searching back through the commit history to see if `total`'s declaration was somehow deleted, causing the bug. I used: 

```git log -Gtotal components/sys/drivers/net.c```

And it was added at the one commit and never changed.

the0ne, 2022-05-24:
Oh, that's sad.
This probably has to do with the discussion around https://github.com/whitecatboard/Lua-RTOS-ESP32/pull/238 that has lead to not merging it yet.
Despite that, it shouldn't have happened of cause.

Please try to cherry-pick/merge the corresponding commits into your source tree. Will it build afterwards?

the0ne, 2022-12-02:
Fixed as 6b455df4965c2242754a80896562e254ba0ee528 - closing.
Thanks for reporting the issue!

closed 2022-12-02 by the0ne