whitecatboard/Lua-RTOS-ESP32 #414: Xtensa syntax error on build

sethpeace, 2022-07-02:
Building gives this output:
```
Using  sdkconfig file
Finding a serial port with USB VID:PID SETH-OTA ...
No serial ports defined in boards.json, using the serial port defined in ESPTOOLPY_PORT
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: D: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: DP,b]]P]PTT@T@DDPtd$$N$NQtdRtd: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: DP: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: ELFD@@r@8: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 6: Syntax error: word unexpected (expecting ")")
expr: syntax error: unexpected argument ‘5.2.0’
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
Toolchain path: /usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: D: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: DP,b]]P]PTT@T@DDPtd$$N$NQtdRtd: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: DP: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: ELFD@@r@8: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 6: Syntax error: word unexpected (expecting ")")
expr: syntax error: unexpected argument ‘5.2.0’
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: D: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: DP,b]]P]PTT@T@DDPtd$$N$NQtdRtd: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: DP: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: ELFD@@r@8: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 6: Syntax error: word unexpected (expecting ")")
expr: syntax error: unexpected argument ‘5.2.0’
CC build/bootloader/bootloader_support/src/bootloader_flash.o
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: D: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: DP,b]]P]PTT@T@DDPtd$$N$NQtdRtd: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: DP: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: ELFD@@r@8: not found
/usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 6: Syntax error: word unexpected (expecting ")")
make[2]: *** [/root/esp-idf/make/component_wrapper.mk:290: src/bootloader_flash.o] Error 2
make[1]: *** [/root/esp-idf/make/project.mk:578: component-bootloader_support-build] Error 2
make: *** [/root/esp-idf/components/bootloader/Makefile.projbuild:41: /root/Lua-RTOS-ESP32/build/bootloader/bootloader.bin] Error 2
```
How do I fix it?

the0ne, 2022-07-02:
Using the right ESP-IDF version?

sethpeace, 2022-07-02:
Just the latest right?
[image]
If so, yes I am using the latest version

the0ne, 2022-07-03:
No. Please see e.g. #350

sethpeace, 2022-07-03:
following those directions still gives the same error

the0ne, 2022-07-03:
Sorry to hear that 🤷🏼‍♂️

sethpeace, 2022-07-03:
You should provide a docker image so building this project isnt so complicated

the0ne, 2022-07-03:
Feel free to create a PR 🙂👍🏻

sethpeace, 2022-07-03:
If I could get it to build I would lol

sethpeace, 2022-08-31:
works now

closed 2022-08-31 by sethpeace

the0ne, 2022-11-14:
There's concrete plans to upgrade the ESP-IDF to the latest version.
I just realized you have already done that?! That's fantastic!
Would you create a [PR](https://github.com/whitecatboard/Lua-RTOS-ESP32/pulls) so all can profit from your work?

TotallyNotSethP, 2022-11-14:
Yeah I'm currently trying to replicate what I did so I can pr it in so stay tuned!

the0ne, 2022-11-14:
😃👍🏻