whitecatboard/Lua-RTOS-ESP32 #418: Running gdisplay.attach in system.lua or autorun.lua puts garbage on top rows of tft screen

sethpeace, 2022-08-30:
[image]

This doesn't occur when running `gdisplay.attach` in a regular file.

sethpeace, 2022-08-30:
Also you can't clear the garbage unless you reboot the system

sethpeace, 2022-08-30:
Ok a temporary workaround is to run `gdisplay.attach` in a regular file, then `dofile` from `system.lua`

sethpeace, 2022-08-31:
Update: it now shows garbage no matter when you run the attach command...

sethpeace, 2022-08-31:
This might be the same issue as https://forum.lvgl.io/t/random-colored-lines-in-display-gc9a01/4545

sethpeace, 2022-08-31:
It works fine from the command line

sethpeace, 2022-08-31:
Oh, and it works now... very inconsistent behaviour

sethpeace, 2022-08-31:
In a shocking new development, i cant put anything in the top two rows anymore. {0,0} is {0,2} on the actual screen

sethpeace, 2022-08-31:
I suspect this may just be an issue with the buffer being two rows down from where it should be

sethpeace, 2022-09-03:
@the0ne Do you know where the interface code is for the ST7735 1.44" green tab... I want to try to fix this issue

the0ne, 2022-09-05:
That would be
```
components/sys/drivers/st7735.c
components/sys/drivers/st7735.h
```
I guess.

the0ne, 2022-12-02:
@TotallyNotSethP did you finally find a solution?
Feel free to reopen in case of questions or create a PR if you want to share your solution with others.

closed 2022-12-02 by the0ne