whitecatboard/Lua-RTOS-ESP32 #419: No way to specify where to keep the history file

sethpeace, 2022-08-30:
I need to be able to tell the os to always put the history file in ramfs, never in fat storage. This is not an option, it always defaults to fat storage.

the0ne, 2022-08-30:
Using df I think it's possible to show the used/free space for each mounted file system.
Maybe you can use that to check if the ramfs has been mounted.
If not, you could add an option to the shell.history() call to make sure it's only activated if ramfs is properly mounted.

sethpeace, 2022-08-30:
Sorry I think you misunderstood... if both ramfs and fat are mounted, i want to be able to choose where the history file is stored. atm, it defaults to fat always

the0ne, 2022-08-31:
It all comes down to function `mount_history_file` in components/sys/sys/mount.c
It **first** checks for fat, then **second** it checks for ramfs.
```
    if (mount_is_mounted("fat")) {
        path = mount_get_mount_path("fat");
    } else if (mount_is_mounted("ramfs")) {
        path = mount_get_mount_path("ramfs");
    }
```

That's clearly because it reflects the idea of a history best: To preserve previous commands.
Which, one has to admit, is better done on a FAT than on a RAMFS 😉

I'd say you could add an optional parameter to function `os_history` in loslib_adds.inc
That parameter could say whether RAMFS should be preferred over FAT - with the default being the current behaviour.
Then in mount_history_file that parameter could be used to check for RAMFS before FAT.