whitecatboard/Lua-RTOS-ESP32 #420: Docker build crashes instantly

sethpeace, 2022-09-24:
When I run `docker build -t luartos:1.0.0 .`, it gets to Step 10 then complains:
```
Step 10/15 : RUN /usr/bin/python -m pip install --user -r ~/esp-idf/requirements.txt
 ---> Running in bb6e9b3f780e
Could not open requirements file: [Errno 2] No such file or directory: '/home/builder/esp-idf/requirements.txt'
The command '/bin/sh -c /usr/bin/python -m pip install --user -r ~/esp-idf/requirements.txt' returned a non-zero code: 1
```
What's going on?

elsbiet, 2022-10-07:
sad to say: i'm afraid, lua-rtos-esp32 is dead.

the0ne, 2022-10-07:
@elsbiet Not sure your comment is any good. You could help instead. It's a community project so everyone should feel invited to help each other.

progerstar, 2023-03-01:
You can change this line in the Dockerfile to RUN /bin/sh -c /usr/bin/python -m pip install --user pyserial future cryptography pyparsing

aplimg, 2023-03-24:
@progerstar, thank you for the line change information of the Dockerfile. It works.