whitecatboard/Lua-RTOS-ESP32 #423: Problems connecting to the Wifi AP from newish devices

xopxe, 2023-07-26:
We are receiving reports of people having problems connecting to ESP32-Things Wifi in AP mode. Same people were using it successfully with previous devices (mostly Android/iOS tablets/phones and laptops).
The failures appear in several way: timeouts while connecting, "bad password" error, or connecting but then failing to acquire an IP address through DHCP. It seems to happen to very new Android and Apple phones. Also, the network is reported as having "weak security", tough it is the default WPA_WPA2 mode.
Any idea about where to look?

the0ne, 2023-08-09:
Really strange. WPA2 shouldn't be reported as weak security.

the0ne, 2023-08-09:
I can't reproduce the problem.
I'm using the following script to start the AP (and the captive portal)

```
net.wf.setup(net.wf.mode.AP, "ESP32 Setup", "", net.wf.powersave.MODEM)

-- Start wifi
try(
        function()
                output = net.wf.start()
        end,
        function(where, line, error, message)
                print(message)
        end,
        function()
                -- Network stat
                net.stat()

                -- Start the HTTP Server
                net.service.http.stop()
                net.service.http.start()

                -- Start the CAPTIVE DSN Server
                net.service.captivedns.start()
        end
)
```

Didn't show as weak security in Android nor in Windows. Didn't receive a timeout when connecting. Didn't have problems with receiving an IP. Didn't have problems when opening the captive portal page.

Maybe you could confirm you have enough free memory and not too many devices connect in parallel.

xopxe, 2023-08-10:
Thanks for the verification.
In your example there's no password set, so it is WEP, no?
I verified the "weak security", it's because components/sys/drivers/wifi.c:343 uses WIFI_AUTH_WPA_WPA2_PSK which allows WPA. Changed the constant to WIFI_AUTH_WPA2_PSK to force wpa2 and my Android 13 started to report it as "strong". 
Testing from the master with this everything seems to work. But I do see the problem with our code deployed, so they can be memory problems, indeed, will look into that.

the0ne, 2023-08-11:
That sounds absolutely reasonable, thanks for pointing it out 😎👍🏻
I'll verify it as soon as possible.

xopxe, 2023-08-14:
Just to keep you informed, we found a stack overflow in our code just as the network is being setup (using smashing protection mode). We are testing increasing the stack, I hope it's just that.

the0ne, 2023-08-21:
Hi @xopxe I just checked our build and it seems we've stumbled across this already.
We didn't have time yet to further test and finally push this. 😢

`wifi_config.ap.authmode = (*password ? WIFI_AUTH_WPA2_PSK : WIFI_AUTH_OPEN); //don't use WIFI_AUTH_WPA_WPA2_PSK here or iOS will go MAD`

The main reason we couldn't do it yet is that we don't yet use the WiFi in AP mode. 🤷‍♂️
I can definitely confirm that going with WIFI_AUTH_WPA2_PSK you're on the right track 👍🏻

the0ne, 2023-08-21:
On line 343 in components/sys/drivers/wifi.c please use WIFI_AUTH_WPA2_PSK instead of WIFI_AUTH_WPA_WPA2_PSK.

closed 2023-08-21 by the0ne