whitecatboard/Lua-RTOS-ESP32 #424: Reusing net.en.start()

progerstar, 2023-08-16:
DHCP after DHCP

/ > net.en.setup()
/ > net.en.start(false)
[..IP is ok..]
/ > net.en.stop()
/ > net.en.setup()
/ > net.en.start(false)
stdin:1: 503316483:can't connect check cable
stack traceback:
	[C]: in field 'start'
	stdin:1: in main chunk
	[C]: in ?

Static after DHCP failed

/ > net.en.setup()
[..DHCP server is unpulgged..]
/ > net.en.start(false)
stdin:1: 503316483:can't connect check cable
stack traceback:
	[C]: in field 'start'
	stdin:1: in main chunk
	[C]: in ?
/ > net.en.setup( net.packip(192,168,1,200), net.packip(255,255,255,0), net.packip(192,168,1,1), net.packip(8,8,8,8), net.packip(8,8,4,4) )
/ > net.en.start(false)
/ > net.stat()
...
ip address 0.0.0.0 netmask 0.0.0.0
gw address 0.0.0.0

the0ne, 2023-09-19:
Confirmed.
Did you look into the code to find the root cause?

progerstar, 2023-09-22:
Yep, net.en.setup() doen't run twise, because status_get(STATUS_ETH_SETUP) still true after deinit.
The fix is an additional command status_set(0x00000000, STATUS_ETH_SETUP); in a *eth_stop().

the0ne, 2023-09-22:
Sure that's really enough? I think I had tested that before and it wasn't sufficient.