whitecatboard/Lua-RTOS-ESP32 #426: mkromfs segfault

progerstar, 2023-09-21:
If there is more than one file in a mkromfs src folder, I get a segfault error in a "romfs_file_seek_internal" function due to access to "entry->file.content".
Am I violating any rules for creating the src folder?

the0ne, 2023-09-24:
The [Wiki](https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/File-systems) states about the ROMFS:

> ROMFS is a file system, developed by the Lua RTOS team from the scratch, in which all it's data is stored together with the Lua RTOS firmware.
> 
> Although you can use other general-purpose file systems, such as SPIFFS or LFS (and mount them as read-only), the use of ROMFS has the following advantages:
> 
>     There is no wasted space, since in ROMFS each file consists of a single block of data, that has the same size as the file content.
>     As the file system is builded and linked together with the firmware, it can be updated through OTA.
>     Small footprint, and minimal RAM usage (usually 1K per opened file).
>     ESP32 firmwares based on Lua RTOS and Lua scripts are deployed in the same way as firmwares enterely written in C.

As it says _[...] each file consists of [...]_ the issue can't be that you include more than one file.
I've never tried to use the ROMFS so I can't give more advise than this.

progerstar, 2023-09-24:
I've fixed it.

I've found a first strange thing in an `add_entry()` function
`memcpy(pa_new_entry->name, name, name_len); `
where name_len - filename length, but pa_new_entry->name has only 1 byte size.
so i've changed `romfs_entry_t`'s field char name[1] to name[32]; (I didn't understand the idea, so I'm not sure I didn't break something).

And have changed the following things in the `traverse()` function (I think they depend on the compiler  - mkromfs executes on the pc):
`if (!centry) -> if (centry == NULL)`
and 
`if (!rest) -> if (strlen(rest) == 0) `