whitecatboard/Lua-RTOS-ESP32 #427: load() gives nil

progerstar, 2023-10-09:
load("print(2)") gives nil. why?
and what are the additional parameters: chunkname (default value = "?") and mode = luaL_optstring(L, 3, "bt") ?

try this:
```
	local cmd = "your command here"
	print("now executing "..cmd)
	local function dostring (x) return assert(load(x), "")() end
	print(dostring(cmd))
```

_Originally posted by @the0ne in https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/405#issuecomment-1098922556_

the0ne, 2023-10-10:
I think that's a pretty Lua specific question, maybe best to ask in a forum regarding Lua itself.

progerstar, 2023-10-10:
Generic lua works.
Tests from ./components/fs_images/tests/tests - don't.

progerstar, 2023-10-14:
my mistake

closed 2023-10-14 by progerstar

the0ne, 2023-10-15:
Ok.
What was the cause of the issue?