whitecatboard/Lua-RTOS-ESP32 #430: Can't download firmware when flashing it (certificate signed by unknown authority)

SantaHey, 2025-03-04:
When I run `wcc -p /dev/ttyACM0 -f -d`, I got this output :
```
2025/03/04 14:32:39 AppFolder:  /usr/bin
2025/03/04 14:32:39 AppFileName:  /usr/bin/wcc
2025/03/04 14:32:39 AppDataFolder:  /home/user/.whitecat-create-agent
2025/03/04 14:32:39 AppDataTmpFolder:  /home/user/.whitecat-create-agent/tmp
2025/03/04 14:32:39 connecting to board on /dev/ttyACM0 ...
2025/03/04 14:32:39 attaching board ...
2025/03/04 14:32:39 start inspector ...
2025/03/04 14:32:39 waiting fot ready ...
2025/03/04 14:32:43 detaching board ...
Unknown board model.
Maybe your firmware is corrupted, or you haven't a valid Lua RTOS firmware installed.

Do you want to install a valid firmware now [y/n])? y 

Please, enter your board type:

  1: Whitecat ESP32N1
  2: Whitecat ESP32N1 with OTA
  3: Whitecat ESP32N1 DEVKIT
  4: Whitecat ESP32N1 DEVKIT with OTA
  5: Whitecat ESP32N2 DEVKIT
  6: Whitecat ESP32N2 DEVKIT with OTA
  7: Whitecat ESP32 LORA GW
  8: Whitecat ESP32 LORA GW with OTA
  9: CITILAB ED1
 10: Espressif Systems ESP32-CoreBoard
 11: Espressif Systems ESP32-CoreBoard with OTA
 12: Espressif Systems ESP32 PICO KIT
 13: Espressif Systems ESP32 PICO KIT with OTA
 14: Espressif Systems ESP-WROVER-KIT
 15: Espressif Systems ESP-WROVER-KIT with OTA
 16: SparkFun ESP32 Thing
 17: SparkFun ESP32 Thing with OTA
 18: Adafruit HUZZAH32
 19: Adafruit HUZZAH32 with OTA
 20: Pycom FIPY
 21: Pycom FIPY with OTA
 22: Olimex ESP32-POE
 23: Olimex ESP32-POE with OTA
 24: Olimex ESP32-Gateway
 25: Olimex ESP32-Gateway with OTA
 26: Olimex ESP32-EVB
 27: Olimex ESP32-EVB with OTA
 28: TRAVIS on Olimex ESP32-EVB with OTA
 29: DOIT ESP32 DEVKIT V1
 30: DOIT ESP32 DEVKIT V1 with OTA
 31: WeMos ESP32 with 128x64 OLED
 32: WeMos ESP32 with 128x64 OLED with OTA
 33: EVK-NINA-W
 34: Silicognition wESP32
 35: Silicognition wESP32 with OTA
 36: M5Stack Core Board
 37: M5Stack Core Board with OTA
 38: TTGO Lora32 without OLED
 39: TTGO Lora32 without OLED with OTA
 40: Generic ESP32 board
 41: Generic ESP32 board with OTA

Type: 36
2025/03/04 14:32:46 detaching board ...
2025/03/04 14:32:46 closing serial port ...
2025/03/04 14:32:47 downloading esptool from http://downloads.whitecatboard.org/esptool/esptool-linux.zip ...
2025/03/04 14:32:47 downloaded
2025/03/04 14:32:47 unpacking esptool ...
2025/03/04 14:32:47 downloading firmware from http://whitecatboard.org/firmwarev2.php?firmware=M5STACK...
board upgradeditecatboard.org/firmwarev2.php?firmware=M5STACK: x509: certificate signed by unknown authority
```

I can't flash the firmware, and I can't find any other source to download the firmware manually. Does anyone have any ideas?

the0ne, 2025-03-04:
That server doesn't seem to serve firmware anymore, so you're probably out of luck 🤷🏻‍♂️

SantaHey, 2025-03-04:
@the0ne Is there any way to make the project work today? I'm now trying to build it myself but it seems that the esp-idf structure has changed and is not compatible with this project (but I'm a beginner so maybe I'm making mistakes).

MattBerg1953, 2025-04-11:
Hello for ESP32 Devkit V1 ( number 29) in your list. I used https://dl.espressif.com/dl/package_esp32_index.json in the IDE 2.3.6 preference. Set Baud rate on 115200 when loading